SmokeMonsterPacks/Super-NT-Jailbreak #161: CopySNES SA-1 cart SRAM dumping does not work in v6.9

astro187, 2020-07-05:
Dumping the SRAM for Super Mario RPG results in a .srm file that is all 0's (when viewed with a hex editior).  The ROM dumps successfully and the resulting .sfc file matches the No-Intro verified dump hash (CRC32: 1B8A0625).  After downgrading to v6.6 I was able to successfully dump the same game cartridge and get a valid, populated .srm file that worked as expected with the SD2SNES.  SMRPG is my only SA-1 game cart, but in troubleshooting this issue I found a user who had a similar problem with Kirby Super Star and Kirby's Dream Land 3 (who was also running v6.9) which leads me to believe it is a more general issue with SA-1 games and not specific to SMRPG.  

**Firmware version with issue**: v6.9

**Relevant Super Nt settings**: in all cases, the default settings CopySNES read from the header were used to dump the cart (4MB/32K/SA-1/LoROM).

tjanas, 2021-02-23:
@astro187 Can you retry and see if this is fixed with the latest JB? I don’t see any mention of it being fixed but worth another try.

SmokeMonsterPacks, 2021-02-24:
There will always be a lot of games that can't dump properly from the cart slot, especially special chip games.

astro187, 2021-09-26:
Sorry, this fell off my radar.  I did confirm that this is still an issue with v7.0 shortly after it released.  So, to summarize, the cart ROM and SRAM dumped fine with v6.6, but with v6.9 and v7.0 the ROM dumped ok, but the SRAM did not.  I do not have any other SA-1 chip games, but I would suspect based on other accounts I read online, that is it a problem common to dumping SA-1 games on v6.9 and above.

alexcom84, 2023-07-11:
Should work now with latest fw. Tested Mavelous and Super Mario RPG and the save file was dumper correctly.

tjanas, 2023-07-11:
@astro187 please test with [v7.2](https://github.com/analoguejb/Analogue-Super-Nt-JB/releases/tag/v7.2)