SmokeMonsterPacks/Super-NT-Jailbreak #162: Bahamut Lagoon: dialogue windows lose HDMA gradient after feeding dragons

re4mat, 2021-01-06:
**ROM Name:** Bahamut Lagoon (Japan).sfc
**ROM Checksum:** a98eb5f0521746e6ce6d208591e86d366b6e0479d96474bfff43856fe8cfec12 (sha256sum)
**Environment:** Super Nt JB w/ ROM on internal SD card slot, or Super Nt + SD2SNES/FXPAK (Pro or non-Pro Rev. H)
  - **Super Nt Firmware Version:** 4.9 (official)
  - **Super Nt Firmware Version:** 6.9 (jailbreak)
  - **SD2SNES Firmware Version:** 1.10.3 (official)
  - **SD2SNES Firmware Version:** 1.10.3-frs-v12

## Description
Feeding the dragons will result in the dialogue windows losing the HDMA gradient effect and turning pink. Once the pink window appears, subsequent windows will also exhibit this behaviour until the player enters another dragon feeding screen and exits out of the screen. In Chapter 1, if the player does not reset the gradient by entering a dragon feeding screen and exiting, and immediately enters the house and talks to Matelite, the window will be black, not pink, and will reset to the normal HDMA gradient upon initiating the Princess Yoyo flashback.

This appears to only occur when playing the game from a ROM file; it does not occur on the original Super Famicom cartridge. I've dumped the ROM from the cartridge using CopySNES and confirmed that the ROM's checksum matches the above checksum, which is the same ROM found in the no-intro pack. The bug occurs regardless of whether or not the ROM file is loaded off of an SD2SNES/FXPAK or loaded off of the Super Nt's built-in SD card slot. It may occur with other flash carts, such as the Super Everdrive, but none were available to me at the time of testing.

Initial speculation is that it's a latching error. UnDisbeliever has a pair of HDMAEN latch test ROMs which can be used to determine when the DMA controller latches the HDMAEN register. ([hdmaen_latch_test.sfc and hdmaen_latch_test_2.sfc](https://github.com/undisbeliever/snes-test-roms/releases/tag/v20210106)) If there is a red line it means HDMAEN was written before latching, while if there is no red line then HDMAEN was written after latching. The author did note to me, however, that the first ROM (hdmaen_latch_test.sfc) is not a stable test, saying that "the output is **slightly** different every time you restart or power-cycle the test". (This was my experience as well.) Either way, those test ROMs may be useful for investigation of this issue, which is why I bring them up here.

## Steps to Reproduce
1. Load from save. (Save file attached below.)
2. Once you are able to move, talk to Zora (who is standing in front of the blue dragon) and receive various grasses.
3. Feed Salamander (the red dragon) 1 Flame Grass (ほのおのくさ).
4. Feed Ice Dragon (the blue dragon) 1 Ice Grass (こおいのくさ).
5. Feed Molten (the green-winged dragon) 1 Life Grass (いのちのくさ).
6. Repeat steps 3–5 until the bug occurs.
NOTE: The number of times the dragons need to be fed appears to be random, and the order in which they are fed does not matter. (What item(s) they are fed probably doesn't matter, but I've only ever fed them their appropriate grasses.) To increase the number of times the dragons are fed and thus increase the chances of the bug triggering, you may wish to name Thunderhawk (when on the naming screen in Japanese, the right-most command confirms the name) and include him in the cycle, feeding him Thunder Grass (いかずちのくさ). While the bug will normally trigger quickly, in very rare cases you may need to be patient and keep attempting to reproduce.

## Expected Result
Dialogue window backgrounds should always have the HDMA gradient effect applied.

## Screenshots

Normal/expected window behaviour, showing the HDMA gradient effect:
[image]

Bugged window behaviour:
[image]

## Super Nt Settings
The settings don't seem to matter, as I tried various permutations to see if any would solve the issue, but these are the current settings I use:

**Video**
Advanced Mode: [X]
Resolution: 1080x60
Width & Height
— Width: +1302
— Horizontal Position: +32
— Height: +1140
— Vertical Position: +44
Cropping
— Crop Left: +0
— Crop Right: +0
— Crop Top: +0
— Crop Bottom: +0
Scalers
— No Scaler: [X]
— Disable Interpolation: []
Scanlines
— No Scanlines: [X]
— Scanline Depth: +120
Extra Features
— 64 Sprite Tiles: [X]
— Mode 5 & 6 Interlace Disable: []
— Pseudo-Hires Blending: [X]
Buffer Mode
— Zero Delay: [X]

**Audio**
-3db Output Cut: []
Cartridge Audio Enable: []
Swap Left & Right: []
Cartridge Audio Volume: +175

**System**
Menu Options
— Skin: SF
— Highlight Style: Solid
— Highlight Color: Red
— Font: Alternate Font, Mixed Case
— Menu Bounce: [X]
— Dim Game in Menu: [X]
Controls & Hotkeys
— Menu and Reset hotkeys are disabled before running the game
— Confirm/Select Key: A
— Cancel/Back Key: B
Startup Options
— Boot Sequence Delay: +4
— Title -> Menu: [X]
Hardware
— NTSC: [X]
— SD Card Speed: Faster
LED Options
— SF: [X]
— LED Brightness: +2
— Adjust Pattern
———— Animate Pattern: []
———— Pattern Position: +0

## SD2SNES Settings
The settings don't seem to matter, as I tried various permutations to see if any would solve the issue, but these are the current settings I use:

**In-game Settings**
Start with cheats enabled: No
Reset to menu: On
In-game hook: Off
Auto region patch: Off
1CHIP transient fixes: Off
Brightness limit: Off
Reset patch for clock phase: Off

**SuperCIC Settings**
Enable SuperCIC: No

## Attachments
Save file: [Bahamut Lagoon (Japan).srm.zip](https://github.com/SmokeMonsterPacks/Super-NT-Jailbreak/files/5773423/Bahamut.Lagoon.Japan.srm.zip)

tjanas, 2021-02-23:
Can you retest with JB v7.0?

re4mat, 2021-02-23:
Sure thing! I'm testing on the official 5.0 as we speak. I'll re-flash to JB 7.0 once this is done and update the bug once both are confirmed fixed.

re4mat, 2021-02-24:
Environment: Super Nt JB w/ ROM on internal SD card slot, or Super Nt + SD2SNES (non-Pro Rev. H)

Super Nt Firmware Version: 5.0 (official)
Super Nt Firmware Version: 7.0 (jailbreak)
SD2SNES Firmware Version: 1.10.3 (official)

For each environment, where applicable, I ran through the repro steps until food ran out 5 times on the Japanese ROM (crc32: 1b83c440), as well as Near's English translated ROM v1.1 (crc32: e17df460, which in my testing would trigger the bug more often than both the Japanese ROM and earlier versions of the translation). Also ran through the repro steps until food ran out 5 times on the original Japanese cartridge as a regression test, both on the official 5.0 and jailbreak 7.0.

Results: After approximately 13 hours of testing, bug appears to be fixed with no regressions found. Closing issue as fixed.

closed 2021-02-24 by re4mat