SmokeMonsterPacks/Super-NT-Jailbreak #168: Power Soukoban (Jap) Graphics Corruption

Chronokeys, 2021-02-24:
**This graphic corruption applies to the rom played through the sd card slot. The japanese rom and its translation are both affected.**
Choosing "continue" after been killed causes the graphics to appear garbled or corrupted.
It's very easy to repro even on the Japanese rom.
## Expected behavior and actual behavior

**Expected Behavior:**
No Garbled Graphics
[image]

**Actual Behavior:**
Graphics are corrupted
[image]

## Steps to reproduce the problem
If you use a translation patch just go into the game, save and take hits until you die, when you hit continue graphics will be messed up. 
Here are the steps for the Japanese rom:
1. After the "press start" screen choose the following options:
    Top choice
2. Move character towards the crack on the upper right side of the wall, face the crack and hold A to break the wall, move ahead.
3. Push the block into the hole and continue through the left exit
4. Here you need to save your game, this are the steps:
     Push select to enter the menu, then "A", select the bottom option, select top option, game is now saved.
5. Exit the menu and wait until enemies crawl out the hole on the left and let yourself die by their hand. It takes two hits.
6. When given the option choose the top one to continue.
7. Success! Graphics are now garbled and will continue to be even if you change rooms
[image]

## ROM name and hash value

**ROM:** Power Soukoban (Japan).sfc
**CRC32:** 39fbaa1e

## firmware

- firmware version: Latest JB v7.0

## relevant Super Nt settings
Stock SNT settings

GreatHierophant, 2021-05-25:
I can reproduce this bug with an sd2snes and the Super Nt.

tjanas, 2023-05-05:
@GreatHierophant can you reproduce this issue with the latest official firmware and sd2snes?

closed 2023-07-11 by tjanas