SmokeMonsterPacks/Super-NT-Jailbreak #173: Suggestion: GBA game suppoort

RealEMK, 2021-04-19:
I'm pretty new to FPGA emulation, but from what I understand it allows you to emulate the hardware of specific consoles for more accurate emulation than software emulation.

My question is, is it possible to implement a mod that allows you to boot into some sort of GBA mode that lets you boot up GBA rom files and changes the hardware emulation to match .gb and .gba files?

RealEMK, 2021-04-19:
Sorry in advanced if I didn't use the correct flag. New to Github and I'm still learning to use the interface properly.

tjanas, 2021-04-30:
I don't think this is going to happen and seems to be out of scope for the Super Nt.

closed 2021-04-30 by tjanas

RealEMK, 2021-05-10:
Any reason in particular you think this is out of it's scope?