SmokeMonsterPacks/Super-NT-Jailbreak #174: Save instructions

cmcavoy, 2021-04-19:
The README says, "When a game is exited to the menu, it will prompt you to save. Return to the file menu and save your progress before powering down." That option was turned off in settings. To get the prompt, I had to turn it on. Would suggest either turning it on as the default, or including the need to turn it on in the README.

Otherwise, this firmware is fantastic...thanks a bunch

tjanas, 2021-05-08:
Submit a pull request with your documentation change.

cmcavoy, 2021-05-08:
@tjanas added: https://github.com/SmokeMonsterPacks/Super-NT-Jailbreak/pull/183

Thanks!

tjanas, 2021-05-08:
Fixed via https://github.com/SmokeMonsterPacks/Super-NT-Jailbreak/pull/183

closed 2021-05-08 by tjanas