SmokeMonsterPacks/Super-NT-Jailbreak #176: Fire Emblem 5: Thracia 776 - Lil' Manster Translation Patch

nmacneil, 2021-04-21:
## Expected behavior and actual behavior
Link to patch: https://forums.serenesforest.net/index.php?/topic/90189-fe5-lil-manster-pe-based-english-translation-translations-to-es-fr-de-it-tr/

While it is recognized that the Lil' Manster patch of FE5 is not compatible with all emulators due to the ROM size being changed, it does not use any special chips. This is one of the only ways to play this Fire Emblem title in English.

Currently, this patched ROM does not run on the Super NT JB. It does work fine using an SD2SNES.

Would it be possible to make this patched ROM playable on Super NT JB?
Is this something that has to be done from the modder's side?

## Steps to reproduce the problem
1. Follow the instructions in the Lil' Manster "Patching instructions.txt" file.
2. Load the patched ROM onto an SD card.
3. Try to boot it using Super NT JB.

## firmware
- firmware version: Super Nt Jailbreak v7.0

### Thank you

nmacneil, 2021-04-21:
It was recently brought to my attention that SD2SNES had a memory map added for this rom by redguy to accommodate this ROM, pre firmware update.

https://krikzz.com/forum/index.php?topic=7230.0

tjanas, 2023-07-11:
@nmacneil please retest with [JB v7.2](https://github.com/analoguejb/Analogue-Super-Nt-JB). It should be fixed now per the change log.

closed 2023-07-11 by tjanas