SmokeMonsterPacks/Super-NT-Jailbreak #177: Push game saves to Cartridge (Feature Request)

haincha, 2021-04-21:
Hello, I am not sure if there is any limitations to the Super Nt hardware that would prevent this from happening, and I couldn't find any direct ways of reaching out.

Could it be possible to push a game save from the SD card to an actual cart? I've seen other hardware with the capability (Redtrode 2), but not seen anything around the Super Nt. I've got some game saves from emulators that I'd like to push to my legit physical carts. I know I could order an FXPAK Pro and just run from that. But, I've got a legitimate collection of games that I'd like to utilize now that I was able to get ahold of a Super Nt.

So just asking:
A is this even possible?
B is this something you'd consider adding in the jailbreak firmware?

Thanks!

PizzaCube, 2021-05-23:
ya i need this on all the consoles  mega sg nt mini noir super nt  to back up save games after i change battery

closed 2023-05-05 by tjanas

tjanas, 2023-07-11:
@haincha @PizzaCube please retest with [JB v7.2](https://github.com/analoguejb/Analogue-Super-Nt-JB). This feature has been added.