SmokeMonsterPacks/Super-NT-Jailbreak #179: Saving feature does not seem to be working

metamorphling, 2021-04-26:
## Expected behavior and actual behavior
>When a game is exited to the menu, it will prompt you to save.
It does not.
Either it's a misunderstanding on my part and wiki is not thorough about this, or it's a bug.

## Steps to reproduce the problem
- Start a game from SD card;
- At any time during gameplay press SELECT + Down;
- File browser menu opens, no save popup is shown;
- Leaving file browser / switching to options - still no save screen;
- SAVES/SNES is empty;

## ROM name and hash value
Any game of any region, I'm afraid it's not related.
Tested with SD card only though.

## firmware
- firmware version: 7.0

## relevant Super Nt settings
- Settings -> System -> Menu Options -> Prompt when save RAM (ON)

tjanas, 2021-04-30:
Does the SNES/SFC icon briefly show up in the lower-right corner when pulling up the menu after playing a game that uses SRAM?

metamorphling, 2021-05-01:
>a game that uses SRAM
alright, now I see how it works. All the games I've tried were games without writeable memory.
Usually anybody who have ever used emulator may assume that "save" goes like save state as in full memory dump, my mistake.

If it's possible can any clarification be added to readme about this? 
Because currently it does not state prerequisites needed for save to work(as in game should have internal save mechanism in the first place), which may be obvious for some users but still feels like a vital detail.

tjanas, 2021-05-01:
I think the readme is clear; it goes into great detail about the cartridge dumping functionality of CopySNES, and it also explains where roms loaded via JB read/write their save data (and how it even will fallback to look in the same directory structure as sd2snes does).

closed 2021-05-01 by tjanas