SmokeMonsterPacks/Super-NT-Jailbreak #180: CopySNES Stunt race fx (Pal) save game

Ramzanox, 2021-04-30:
CopySNES on Stunt race fx (Pal version) copys the game and save game, but only the game works but not the save game.

ssshake, 2021-05-01:
I'm curious about this, I don't understand. This is a SFX chip game so how is it that you are saying the game works after backup? It shouldn't work at all right?

alexcom84, 2022-11-01:
What this user means is SRAM dumping isn't working. It's broken since fw 6.9 and downgrading to 6.6 works as intended. 
 Reproduced this issue with SA-1 games and SPC7110 Far East of Eden (should affect all special chip games).

tjanas, 2023-07-11:
@Ramzanox can you please re-test with [JB7.2](https://github.com/analoguejb/Analogue-Super-Nt-JB/releases/tag/v7.2)?

tjanas, 2023-07-11:
If issue still exists, please report the issue at the [analoguejb](https://github.com/analoguejb/Analogue-Super-Nt-JB) repository.

closed 2023-07-11 by tjanas

Ramzanox, 2023-07-12:
> Lähetetty: tiistai 11. heinäkuuta 2023 21.56
> Vastaanottaja: SmokeMonsterPacks/Super-NT-Jailbreak ***@***.***>
> Kopio: Ramzanox ***@***.***>; Mention ***@***.***>
>
> Closed #180<https://github.com/SmokeMonsterPacks/Super-NT-Jailbreak/issues/180> as completed.

Tested today copysnes on stunt race fx (pal version) with JB7.2 and sadly the save doesnt still work when i tested it on computer.

Ramzanox, 2024-08-20:
Tested today with retrode 2 to dump game+savegame and seems atleast on emulators that the savegame doesnt work, on real snes console and Super Nt my game works.