SmokeMonsterPacks/Super-NT-Jailbreak #181: Entering "Browse SD Card" while running from Cartridge forces you to restart the game

ssshake, 2021-04-30:
## Expected behavior and actual behavior

When playing a came from cartridge if you enter the Browse SD Card menu, upon hitting "back" it lets you get out of the on screen menu and continue playing the cartridge came.

Actual behavior is that it acts the same as when you've loaded a came from ROM. It says "You are about to leave the file browser. Doing so will stop the running game..."

There doesn't seem to be any way to dismiss the on screen display and continue the game. 

This is a big problem because if you are in the middle play playing an RPG from cartridge, if you wander into the SD browser menu by accident, you will be forced to reset the game and you could lose some significant progress.

## Steps to reproduce the problem

Load a game from cartridge
Open the on screen menu pressing select + down
Go into SD Browser
Attempt to Exist SD browser or press select to go "Back to Game" according to the OSD.
Neither work. 

## firmware

- firmware version: v7.0

tjanas, 2021-05-01:
[image]
I think you are mistaken. I just tested this; press B to go back to file browser (game still running), and then press SELECT to go back to your game as you left it.

ssshake, 2021-05-01:
I'm quite positive what I'm saying is true and accurate. I don't know why you can't reproduce it, so maybe we're both right?

I only just received this SuperNT yesterday. 

Here is a video where I show the issue. [https://youtu.be/ebfjSIW1bWY](https://youtu.be/ebfjSIW1bWY)

ssshake, 2021-05-01:
I think the distilled issue is this:

IF a game was loaded from rom, when in the SD browser, pressing select exits the browser and resumes the game.
IF a game was loaded from cartridge, when in the SD browser, pressing select does not exit the browser.

It seems to be that the SD browser can only be exited if currently running game was from the SD card.

tjanas, 2021-05-08:
I have confirmed this issue.

tjanas, 2023-07-12:
With JB 7.2, entering Browse SD Card while running a cartridge or the built-in Super Turrican will immediately stop running the cartridge & game.