SmokeMonsterPacks/Super-NT-Jailbreak #187: SNES PowerPak compatibility

Ramsis-SNES, 2021-05-13:
## Summary
The SNES PowerPak flashcart by RetroUSB seems to be incompatible with the Super Nt. Tested with firmware JB7.0 on a brand-new Super Nt ordered directly from Analogue in early April, two different PowerPaks (revD and revH) and two different CF cards (SanDisk 4 GB and Dane-Elec 4 GB).

N.B. Although the problem was brought up on [Twitter](https://twitter.com/sst85official/status/962726374149562368) quite some time ago, I couldn't find a working solution (sorry if I missed it), or even an existing GitHub issue, so here we go.

## Description
### Expected behavior
Well, it should work like any other flashcart. ;-) (Regular game cartridges work just fine on the console BTW, as does my sd2snes classic.)

### Actual behavior
When booting the PowerPak with [my custom firmware](https://github.com/Ramsis-SNES/snes-powerpak-firmware-v3) installed, I get a screen with garbled graphics:

[image]

Note that I can still "blindly" control the menu (e.g., access the settings and whatnot), but trying to load an actual game always results in either a freeze (card error message?) or a black screen.

Without a CF card inserted, at least the graphics load fine:

[image]

I did downgrade one of the PowerPaks to its stock firmware (boot ROM v1.02, card ROM files v1.05). Menu graphics load fine there, but games don't (it complains about a missing internal ROM header, just like "Steven" and "Chris" on Twitter mentioned). 

## Steps to reproduce the problem
Boot your SNES PowerPak on the Super Nt and try loading/launching a game from the CF card.

## Relevant Super Nt settings
* The "Launch system timing" setting has no effect on this whatsoever. I tried not only changing the setting and "warm-booting" the PowerPak via the Super Nt menu, but also saving the Super Nt settings and then power-cycling the console/"cold-booting" the cart. No change in behavior. (Note that the PowerPak does work flawlessly on newer consoles such as 1CHIP and mini/Jr. ones, so any slight timing differences between those and the SHVC-CPU-01 model actually shouldn't even matter.)
* I tried it all with and without an SD card inserted into the Super Nt, no change in behavior.
* I tried it all both with the OEM power supply (with a voltage step-down transformer-adapter as I live in Europe) and with a European USB power supply that can provide even more power than required by the Super Nt (2.4A), no change in behavior.

## Thoughts and facts
Games not wanting to load on the one hand (stock firmware) and garbled menu graphics on the other (my firmware, which loads all menu graphics from a theme file on the CF card) both seem to suggest that the Super Nt somehow disrupts data transfers from CF to SNES WRAM/PowerPak SDRAM. (Yes, the PowerPak hardware is designed to route all CF data through the SNES CPU ... don't ask me why).

Then again, the PowerPak obviously succeeds in loading some critical data from the CF card (system configuration, FPGA bit file), which happens _before_ the start screen appears (stock firmware) or theme file data is loaded (my firmware), respectively. Of course, it is unclear at this point if the data received is sane at all. With the stock firmware, at least the contents of the root directory seemed to be represented just fine.

The PowerPak uses DMA for most of its CF data transfers via a custom port (`$21FF`) even if the user decides to "disable" DMA (which is possible), though I may have inadvertently made changes to the way the setting is acknowledged. (Also, it's been a while since I worked on the firmware at all. In any case, the CF software interface is still a complete mess even after all these years, but who cares ... :p) Also, the PowerPak is known to manipulate the `$21CX` SNES bus range for no good reason – Ultra16 owners will remember the hardware errors this used to produce on their consoles.

That's it for now. It's after 2 am, and I'm tired. XD

So long, and thanks for reading!

closed 2021-05-13 by tjanas

Ramsis-SNES, 2021-05-13:
Welp, thanks for nothing.

tjanas, 2021-05-13:
This cart offers zero value over using the jailbreak firmware, and possibly supports less than the jailbreak.

Ramsis-SNES, 2021-05-13:
That's debatable (PowerPak has e.g. an SPC auto-play feature that automatically cycles through a directory), but it should work regardless, especially since Analogue's [Super Nt Reference Manual](https://support.analogue.co/hc/en-us/article_attachments/360066166691/Analogue_Super_Nt_Reference_Manual.pdf) explicitly lists it under compatible cartridges (p. 7).

lantisescudo, 2021-05-14:
According to kevtris on the Classic Gaming Discord, "the snes powerpak should work if the "original system timing" checkbox is checked in the hardware menu"

Ramsis-SNES, 2021-05-15:
Thanks, but as I mentioned, that setting doesn't seem to make any difference. To verify again (but in a different way), I pulled my Game Genie out of storage today and tried it on the Super Nt. As I had anticipated, the GG works fine _regardless_ of the setting. I'm starting to wonder whether the Super Nt acknowledges that setting at all, or even if my unit could actually be faulty? Then again, people on Twitter described the exact same problem I have with the PowerPak and _only_ the PowerPak, while everything else I plug into the console works as it should ...

BTW, I tested some more stuff today:
* Downgraded the Super Nt to its latest "stock" firmware (v5.0), and the problem persists. (I didn't actually expect this to fix it but I still needed to know for sure that the issue was present in the regular firmware as well.)
* Temporarily modified my PowerPak firmware to force non-DMA mode for CF --> SDRAM loading operations: No change in behavior _at all_. This is very interesting. Now we at least know what's _not_ causing trouble here. ;-)
* Loaded and started a known working game using the PowerPak's "Play last game" feature (leading to a black screen of course), then used the Super Nt's CopySNES feature to read the game data from the cart's SDRAM and write it back onto the SD card. As it turns out, the game data gets corrupted in such a way that random single bytes are completely missing, shifting chunks of subsequent game data down to lower offsets. This adds up as it goes along, meaning that e.g. the game title (normally at `$7FC0` in this case) is found somewhere in the `$7F20` region.

Sigh. I can't tell you how disappointed I am right now. :-( I was sooo looking forward to using my beloved PowerPak, in glorious HD and with all the Super Nt's nifty video features, on my modern flat TV. Kind of makes me wish I'd spent the money on a RetroTINK 5x or something similar instead ...

Ramsis-SNES, 2022-01-11:
Just so you know, it's a regression in v5.0/JB7.0. The PowerPak works fine with v4.9/JB6.9 ("launch system timing" must be enabled, as was stated above.) I hope kevtris or someone else can do something about it – I'd imagine it's an easy fix.

> As I had anticipated, the GG works fine _regardless_ of the setting.

Small correction: The cart boots up, but entered codes won't have any effect unless it's enabled. So the setting at least _does_ something in v5.0/JB7.0 as well.

tjanas, 2023-07-11:
@Ramsis-SNES please retest with [JB v7.2](https://github.com/analoguejb/Analogue-Super-Nt-JB).