SmokeMonsterPacks/Super-NT-Jailbreak #193: 72hr Kaizo level (Super Mario World SA-1 hack) has loud glitched audio when collecting coins

RaspberryAlpine, 2021-08-22:
On Super Nt Firmwares v4.9 and v5.0 and Jailbreak v7.0
Collecting coins makes a loud glitchy noise 

https://clips.twitch.tv/WittyOpenIcecreamFeelsBadMan

## Expected behavior and actual behavior
- On my SNES Jr + FP Pack Pro 
- and on a Super Famicom (I don't know the revision)+ SD2SNES 
- and on MiSTer SNES core SNES_20210713.rbf
- Super Nt Firmware official v4.7  
- Super Nt Jailbreak v6.6

Collecting coins makes no sound effect. 

## Steps to reproduce the problem
- start the game
- get through the intro text
- on the overworld select the level on the left 
- Collect a coin in the level 

There are coins at the start of the level that are easy to get to.

## ROM name and hash value
31_Fornaxus.sfc 
sha1 - c1d3381a22d6d51775ea007b920319f5fa7a1ee5
md5 - c0b274726d665349cfbee0d875799bb8

Here is a link to the bps patch file. 
https://cdn.discordapp.com/attachments/335174299306688514/879105710665900112/31_Fornaxus.bps

Needs to be patch with 'Super Mario World (USA).sfc'
sha1 - 6b47bb75d16514b6a476aa0c73a683a2a4c18765
md5 - cdd3c8c37322978ca8669b34bc89c804

## firmware
issue occurs on the following firmwares: 
- Official v4.9 
- Official v5.0
- Jailbreak v7.0

Does not occur on:
- official v4.7 
- Jailbreak v6.6

## relevant Super Nt settings
N/A

tjanas, 2023-05-05:
@RaspberryAlpine this has been reported as being fixed in the change log for the official firmware update released today. Please enable SPC RAM startup state in the hardware menu, and re-test with your sd2snes.

tjanas, 2023-07-11:
@RaspberryAlpine please retest with [JB v7.2](https://github.com/analoguejb/Analogue-Super-Nt-JB). Enable SPC RAM startup state in the hardware menu.

closed 2023-07-11 by tjanas