SmokeMonsterPacks/Super-NT-Jailbreak #194: I can't operate the menu

mk7org, 2021-09-04:
Ganbare Goemon Kirakira Dochu Boku ga Dancer ni Natta Wake(Japan) no intro

It is difficult to move to the menu screen during the game.
If you move to the menu screen, it will become inoperable and will black out in the worst case.

Super Nt Firmware official v5.0

Super Nt Jailbreak v7.0

tjanas, 2023-05-05:
@mk7org can you re-test with the latest official firmware update?

tjanas, 2023-07-11:
@mk7org please retest with [JB v7.2](https://github.com/analoguejb/Analogue-Super-Nt-JB).

HitoriSensei, 2023-07-11:
@tjanas Call me paranoid, but the mentioned repo doesn't bring any confidence in term of safety of the firmware. The owner has just created a GitHub account, and these repos aren't even forks. Where did it came from?

tjanas, 2023-07-11:
@HitoriSensei it is the new home of jailbreak firmwares for Nt Mini Noir, Mega Sg, and Super Nt. You can have 100% full confidence that they are just as legitimate as those that were originally hosted here.

HitoriSensei, 2023-07-11:
Well then, I just hope that I won't wake up one day to a ransom paybtc-or-brick message on my SuperNT 😉

tjanas, 2023-07-13:
@mk7org can you provide a video demonstrating the issue you are seeing?