SmokeMonsterPacks/Super-NT-Jailbreak #196: Feature Request: Launch special chip games off SD card using physically present chip in cartridge to boot

teconmoon, 2021-10-02:
This is one of those 2 AM “WHAT IF” ideas, so it’s likely completely flawed.

Currently, games relying on special chips can’t be booted from an SD card due to the limitations discussed in issue #160.

Would it be possible to build a feature into this firmware where games relying on special chips could use the chip from a physically present cartridge? For example, pop in a cheap Japanese SMRPG cartridge, and utilize the cartridge’s SA1 chip to boot Kirby’s Dream Land 3 off an SD card.

Looking forward to any thoughts on this, though I’m sure this won’t be feasible. Thanks for reading!

crudelios, 2021-10-02:
This has been asked in #112 and it can't be done unfortunately.

Please check the detailed answer on that issue.

teconmoon, 2021-10-02:
Gotcha, looks like it’s a cheap 27C322 adapter for me. Thanks for the clarification!

closed 2021-10-02 by teconmoon