SmokeMonsterPacks/Super-NT-Jailbreak #200: "Unsafe" (Speedrunners and the need for a 4th video option)

charlesmadere, 2022-04-13:
Hi everyone,

I think most of us are aware of the Analogue consoles 3 main video settings: **Fully Buffered**, **Single Buffer**, and **Zero Delay**. Each are within the standard featureset of the Super Nt, Mega Sg, and Nt mini Noir. All of these are perfect for HDMI and work great for their use cases, but then also inevitably come down to a compromise.

With **Fully Buffered**, the Analogue console will run at the exact timing as the original console, but with additional input lag in order to meet the HDMI’s standard 60Hz spec.

With **Single Buffer**, the Analogue console will run at the exact timing as the original console, but with variable input lag in order to meet the HDMI’s standard 60Hz spec.

With **Zero Delay**, the Analogue console will either overclock or underclock to meet the HDMI’s standard 60Hz spec. Therefore it has subtly gone out of spec with the original console, but with the bonus of absolutely no input lag.

One could argue, and I think it would be valid, that the above should cover just about any person’s use case. That may be true… but speedrunners need better. Especially for people pushing for top times. Inevitably there are massive compromises with each of the above options.

I speedrun Super Metroid any% on my Super Nt and have a PB of 42:37, a time that puts me at #15 in a game with over 1,000 submitted runs on speedrun.com. And I did this on Zero Delay. Considering that I lose about 1 second every 10 minutes under this setting, does that mean my PB should be about 42:33? Or perhaps I should run on Single Buffer, because then that way I don’t lose time? But that sounds terrible right, to play at high level with variable input lag? So then, maybe I should play at Fully Buffered, to not lose any time at all, and also not have variable lag… but then I’m accepting having even more input lag?

No, this is not why I or any of my friends bought Analogue consoles. We didn’t buy them to have to make compromises like this.

Additionally, the Super Nt and Nt mini Noir are actually advantaged over Mega Sg in this conversation. The Mega Sg actually has to overclock in order for Zero Delay to hit 60Hz, whereas the Super Nt and Noir both downclock. This means that the Mega Sg on Zero Delay should probably be literally banned by any serious speedrun leaderboard.

I would like to put a call out for a 4th video option: “Unsafe”. This would output the console’s true native Hz, HDMI spec be damned, over the HDMI connection. With this option, there is no compromise.

Clearly this introduces a new problem though: what can Analogue do to prevent people from selecting this option, while using hardware that doesn’t support the weird Hz, and then suddenly they have a blank picture on their screen? How does someone recover from this?

I suggest the following solution:

* User navigates to the console’s video modes
* User selects "Unsafe"
* A dialog pops up and says "Warning, your TV may not be able to handle this input. Super Nt will switch to the native console framerate, and then ask you to enter a specific series of button prompts. In 15s if the button prompts have not been entered, Super Nt will fallback to the current video option."
* User confirms
* User then waits for their TV to switch to the different framerate (if it works)
* User then enters the button combo, and they're now good to go with the native Hz
* Alternatively, the user can't see anything on their TV, and after 15s, the Super Nt reverts to the previous video option

\^ The above solution was sent by me to Analogue in December 2021, which a response shortly after:

>Hi Charles, I have passed along your request to our development team. Thank you very much and have a good day! Best Regards, Analogue Support

Now I appreciate this email, it looks great. But given the dearth of general firmware updates, and how freaking strongly I feel about this issue, I am actually seriously considering buying an SNES Jr, getting it RGB modded, buying HD Retrovision cables, and a Retrotink5x, all for a measly \~4s of time save. And you know what, this solution, even though it would work, sucks because (A) it goes against the “perfect” aspect of Analogue consoles (B) it is very expensive (C) a Retrotink5x actually introduces an additional 4ms of input lag.

Thanks so much for your time, please spread this issue around and get the word out if you can.

smCharles

For the nitty gritty on this topic, the following site has all the old consoles and framerates nicely listed and detailed: [https://emulation.gametechwiki.com/index.php/Resolution](https://emulation.gametechwiki.com/index.php/Resolution)

Fellow Super Metroid runners and friends of mine who would also love having this option available to them:

* Imyt
* Eddie
* FreyasSpirit
* CynanMachae

Further conversation on Reddit about this issue: https://www.reddit.com/r/AnalogueInc/comments/u33x4s/unsafe_speedrunners_and_the_need_for_a_4th_video/

helohe, 2022-04-18:
Instead of your proposed solution you might want to look at getting a [MiSTer](https://misterfpga.org/). AFAIK MISTer can run at the native clock and with Freesync and a Freesync compatible monitor that would be the closest you can get to the original hardware.

charlesmadere, 2022-04-18:
@helohe that might be the case for most people, but unfortunately at least when it comes to Super Metroid, and also sort of more broadly just SNES in general, this runs into more problems.

The main problem with Super Metroid being this issue: [SNES_MiSTer issue 296](https://github.com/MiSTer-devel/SNES_MiSTer/issues/296). The MiSTer's SRAM implementation is out of spec for the SNES, which breaks core functionality of the Super Metroid practice rom (obviously this is extremely important for speedrunners).

Then when it comes to just the SNES In general, because the Super Nt has a cartridge slot, it can be used with an FXPAK/SD2SNES, which opens it up to just tons much more cool functionality overall. Be it autosplitters, multiworlds, or other hobbyist things that involve monitoring the game's memory state, none of this is possible on MiSTer. And again, Super Metroid already has a bunch of cool applications that utilize this feature set of the romcart (such as Twitch chat being involved in Phantoon pattern guessing).

tjanas, 2023-07-11:
@charlesmadere if this feature request is still wanted, please consider opening an issue at [analoguejb](https://github.com/analoguejb/Analogue-Super-Nt-JB).

closed 2023-07-11 by tjanas