SmokeMonsterPacks/Super-NT-Jailbreak #202: PowerPak flash card not working on firmware version 5.0 & JB 7.0

dbnhorchner, 2022-08-04:
I ran some tests with my PowerPak and different Super Nt firmware versions. I used the PowerPak (2/23/10, RevE) with the latest official firmware versions (Boot ROM v1.02, Card ROM v1.05, DSP1 installed)

For all the test described below, after installing a specific Super Nt firmware version (5.0, 4.9 and 4.7), I left all settings at default, except for "LAUNCH SYSTEM TIMING" which I enabled.

I tried to start the following games 10 times each from the PowerPak menu (and I rebooted the Super Nt between each attempt):

Super Mario World (USA).sfc [MD5: cdd3c8c37322978ca8669b34bc89c804]
Legend of Zelda, The - A Link to the Past (USA).sfc [MD5: 608c22b8ff930c62dc2de54bcd6eba72]
Super Metroid (Japan, USA) (En,Ja).sfc [MD5: 21f3e98df4780ee1c667b84e57d88675]

These are my results:

**snt_firmware_ver5.0.bin**:

Super Mario World: failed to start 10 out of 10 times
A Link to the Past: failed to start 10 out of 10 times
Super Metroid: failed to start 10 out of 10 times

**snt_firmware_ver4.9.bin**:

Super Mario World: failed to start 4 out of 10 times
A Link to the Past: failed to start 5 out of 10 times
Super Metroid: failed to start 5 out of 10 times

**snt_firmware_ver4.7.bin**:

Super Mario World: started successfully 10 out of 10 times
A Link to the Past: started successfully 10 out of 10 times
Super Metroid: started successfully 10 out of 10 times

When a game fails to start an error message is shown (by the PowerPak) that includes the text "No Internal Header Found" and additional details. The PowerPak works as expected on an official SNES.

Ramsis-SNES, 2022-08-05:
Thanks for bringing this up again (see https://github.com/SmokeMonsterPacks/Super-NT-Jailbreak/issues/187).
It's a real shame that after all this time, the Super Nt still doesn't work as advertised (i.e., supporting flash carts). :-(

BTW, it seems a bit odd that games would fail intermittently using Super Nt firmware v4.9. I did a lot of testing last year and I didn't encounter problems with this firmware and the PowerPak. It does have other severe bugs though (unrelated to the PowerPak) that were fixed in v5.0, so going back again is not an option for me.

dbnhorchner, 2022-08-05:
Yes, I was suprised by the ~50% failure rate on snt_firmware_ver4.9.bin as well. Did you ever try firmware 4.9 with the latest official PowerPak firmware ((Boot ROM v1.02, Card ROM v1.05)? 

I assume this issue is considered low priority because of the jailbreak. But here is one use case why I would like to see proper support for the PowerPak: playing a game on my real SNES, saving and then continue on my Super Nt (and vice versa). The PowerPak might be old and discontinued but it works perfectly fine on a real SNES and it USED to work on the Super Nt, so I think it is fair to expect this to be fixed (eventually).

Ramsis-SNES, 2023-05-04:
@dbnhorchner I tried it with [firmware v5.1](https://www.analogue.co/support/super-nt) that was released just today, and it seems to be fixed. (Make sure you re-enable "Launch system timing" after the update as it will unfortunately reset all settings to their default values.)

Whew, finally. To me, this adds quite a bit of value back to the Super Nt. ^_^ Still feels a bit surreal after all these years. I almost hope it's not just a lucky coincidence. ;-)

closed 2023-05-05 by tjanas

tjanas, 2023-07-11:
@dbnhorchner @Ramsis-SNES please retest with [JB v7.2](https://github.com/analoguejb/Analogue-Super-Nt-JB). Turn on launch system timing in the hardware menu.

horchner-buas, 2023-07-11:
I've retested the PowerPak with JB v7.2 and it works now!