SmokeMonsterPacks/Super-NT-Jailbreak #208: Pocky & Rocky - Flashing screen in 2P while Rocky is shooting.

Seiik, 2023-08-02:
Video of the problem: https://youtu.be/qwzCr4zt0io

I've been trying a couple games with a friend and everything was fine and good, but when playing Pocky & Rocky in 2P, the screen started flashing the moment Rocky was shooting.

I've tried with my original cartridge and with my Retrode-dumped version of it through the SD card. This definitely wasn't happening before the latest firmware as I played it a bunch. Pocky the maiden is 1P Rocky the Racoon is 2P, from that point on, as soon as the game starts, the moment Rocky shoots, the whole screen starts flickering like in the video. 

Setup: 1P: 8bitdo Bluetooth dongle connected to a Switch Online SNES controller.  2P: an OG SNES controller. OG Analogue power supply and cables.

- firmware version: Jailbreak v7.2

Firstly I'd appreciate if someone with a unit could test this, wondering if that's my unit having an issue maybe...

All my display settings are based on Firebrand's OCD settings: https://www.youtube.com/watch?v=MkcayHce6jE

Ricdeau-13, 2023-08-20:
Just updated to latest jailbreak and can confirm that both original NTSC cart and ROM causes screen flicker **_and slowdown_** when moving and shooting as Rocky. Standing still and shooting does not flicker or slowdown. Playing Pocky also runs normal as far as I can tell with no flickering and no immediately eviden slowdown.

Hanskrister79, 2023-08-22:
Send the dump to hanskrister@hotmail .com and I can have a go

Blessings

Norway

Hanskrister79, 2023-08-23:
Playing on latest jailbreak and can confirm the problem

Seiik, 2023-08-24:
Hey, thanks, at least I know it's not my unit being defective. 🙂

Have a nice one!